Xilinx/ml-suite #16: Problem launching instance

rjb4standards, 2018-04-18:
[image]

Your instance configuration is not eligible for the free usage tier
To launch an instance that's eligible for the free usage tier, check your AMI selection, instance type, configuration options, or storage devices. Learn more about free usage tier eligibility and usage restrictions.

I have subscribed to Machine Learning Development Stack from Xilinx, Preview Edition at the $1.65 rate via AWS.
I'm not sure why I'm receiving this error.

rjb4standards, 2018-04-24:
Kamran,

Thanks for your reply, I was able to get past this issue with the help help of AWS support. I have successfully launched an F1.x2 instance and have execute the MXNet tutorial listed here. 

I'm now looking for documentation to train a new MXNet model within the AWS F1.x2 environment. Do you have any suggested reading to train a new model MXNet model within this environment? Thanks.

kamranjk, 2018-05-07:
For training models with MxNet, please refer to the standard MxNet Docs here: https://mxnet.incubator.apache.org/

In the ML Suite, we don't have support for training models, but you will be able to deploy trained models through our xfDNN tools. 

Thanks,
Kamran

closed 2018-06-05 by kamranjk