Xilinx/ml-suite #19: Failed to parse NetParameter file in YOLOv2 Demo App

ericberglund, 2018-07-19:
Following the YOLOv2 demo app on AWS following the instructions [1], the run bash script is failing.
./run.sh aws e2e

I'm seeing the following failure below. If I use the yolov2.caffemodel file found in the YOLOv2 zip file from the download page [2] I get past this failure. It looks like the code is expecting a caffemodel binary file instead of the format found here [3].

I0719 14:52:40.138916  4568 net.cpp:255] Network initialization done.
F0719 14:52:40.139163  4568 upgrade_proto.cpp:95] Check failed: ReadProtoFromBinaryFile(param_file, param) Failed to parse NetParameter file: /home/centos/ml-suite/models/yolov2/caffe/fp32/yolov2.caffemodel
*** Check failure stack trace: ***
./run.sh: line 56:  4568 Aborted                 python yolo.py

[1] https://github.com/Xilinx/ml-suite/tree/master/apps/yolo
[2] https://github.com/Xilinx/ml-suite/blob/master/docs/tutorials/models.md
[3] https://github.com/Xilinx/ml-suite/blob/8f90af4a337a049fae5074c27c7e5e60ad363fc9/models/yolov2/caffe/fp32/yolov2.caffemodel

esberglu, 2018-07-19:
Yep that worked. Thanks for the info!

wilderfield, 2018-07-19:
You are welcome, thanks for opening the issue. I added this info to the
main readme page.

closed 2018-07-20 by kamranjk

adu81020799, 2018-09-21:
I hit the same error again . :<(