Xilinx/ml-suite #20: Failed to find an OpenCL platform YOLOv2 Demo on AWS

esberglu, 2018-07-19:
I'm getting the following error running the yolo demo app on AWS with "./run.sh aws e2e". The compile and quantization steps appear to be completing successfully, xyolo is failing.

* NOTE * Because of https://github.com/Xilinx/ml-suite/issues/19 I am using the yolo_deploy_608.prototxt and yolov2.caffemodel found in the zip file from the model download page, not the versions found in ml-suite/models/yolov2/caffe/fp32/

INFO: Entering XYOLO WITH
Finished batch 1
[XBLAS] # kernels: 1
[0]user:0xf010:0x1d51:[???:??:0]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
[0]user:0xf010:0x1d51:[???:??:0]
device[0].user_instance : 0
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Failed to find an OpenCL platform

wilderfield, 2018-07-19:
Eric,

I'm assuming that you are trying to run on the FPGA Developer AMI.

At the moment, that is the right place to start.

Unfortunately there is a little bit of setup required before you can run.

Do the following:

   1. git clone https://github.com/aws/aws-fpga.git
   2. cd aws-fpga
   3. source sdaccel_setup.sh

esberglu, 2018-07-20:
Thanks again, I was able to complete the yolo demo with this fix and https://github.com/Xilinx/ml-suite/issues/19, with some caveats.

The sdaccel_setup.sh script requires that you are using the 2017.4 version on Xilinx SDx [1] [2]. For now I just changed those to checks to 2017.1, which is the version found on the AWS instance.

The Makefile then was unable to find the libstdc++.so file in the expected location [3]. For now I just searched the filesystem for a libstdc++.so and changed the path in the install command to point that location. After that the sdaccel_setup.sh ran to completion and I was able to finish the yolo demo. However I am not confident that either of those solutions is the "correct" way to be doing things. Perhaps you are aware of a better solution.

[1] https://github.com/aws/aws-fpga/blob/2fdf23ffad944cb94f98d09eed0f34c220c522fe/sdaccel_setup.sh#L160
[2] https://github.com/aws/aws-fpga/blob/2fdf23ffad944cb94f98d09eed0f34c220c522fe/SDAccel/Makefile#L39
[3] https://github.com/aws/aws-fpga/blob/2fdf23ffad944cb94f98d09eed0f34c220c522fe/SDAccel/Makefile#L81

wilderfield, 2018-07-20:
Eric,

When did you start your AMI? The current FPGA Developer AMI should come with the 2017.4 files that play nicely with the latest aws-fpga repo code. The xcllbins included with ml-suite were generated for the 2017.4 dynamic dsa/shell. Looks like they are backward compatible to 2017.1 based on your testing. I think you should consider creating a fresh FPGA developer instance.

esberglu, 2018-07-20:
Ah I was using the Xilinx ML Suite AMI, not the FPGA Developer AMI. Sounds like that's my problem. I'll give it a try with FPGA Developer AMI.

kamranjk, 2018-07-20:
Hi Eric,

sorry about that. We are in the process of updating the AMI to support the latest ML Suite. Should be ready shortly, but as you can start with the FPGA Dev AMI today.

Thanks,
kamran

esberglu, 2018-07-20:
It worked with modifications and no issues with the FPGA Dev AMI. Thanks!

b04902036, 2018-07-25:
I use the Xilinx ML Suite AMI, and I am trying to run the same demo app with the same command "./run.sh aws e2e", I also do the `source sdaccel_setup.sh` after the recomanded modification. However it gives me the following error:
```
[XBLAS] # kernels: 1
[0]user:0xf010:0x1d51:[xocl:2017.4.5:128]
xclProbe found 1 FPGA slots with xocl driver running
CL_PLATFORM_VENDOR Xilinx
CL_PLATFORM_NAME Xilinx
CL_DEVICE_0: 0x3169fe0
CL_DEVICES_FOUND 1, using 0
loading /home/centos/src/project_data/ml-suite/overlaybins/aws/overlay_3.xclbin
AFI not yet loaded, proceed to download.
ERROR: Failed to create compute program from binary -44
```
I also tried the Jupyter tutorial provided by ml-suite, it also gives me the error:
```
loading /home/centos/src/project_data/ml-suite/overlaybins/aws/overlay_3.xclbin
AFI not yet loaded, proceed to download.
ERROR: Failed to create compute program from binary -44
```
Could someone help me out with this?
Sincerely thanks!

closed 2018-08-07 by kamranjk