Xilinx/ml-suite #22: loading AFI error

b04902036, 2018-07-25:
I use the Xilinx ML Suite AMI, and I am trying to run the same demo app with the same command "./run.sh aws e2e" in [this issue](https://github.com/Xilinx/ml-suite/issues/20), I also do the `source sdaccel_setup.sh` after the recomanded modification. However it gives me the following error:
```
[XBLAS] # kernels: 1
[0]user:0xf010:0x1d51:[xocl:2017.4.5:128]
xclProbe found 1 FPGA slots with xocl driver running
CL_PLATFORM_VENDOR Xilinx
CL_PLATFORM_NAME Xilinx
CL_DEVICE_0: 0x3169fe0
CL_DEVICES_FOUND 1, using 0
loading /home/centos/src/project_data/ml-suite/overlaybins/aws/overlay_3.xclbin
AFI not yet loaded, proceed to download.
ERROR: Failed to create compute program from binary -44
```
I also tried the Jupyter tutorial provided by ml-suite, it also gives me the error:
```
loading /home/centos/src/project_data/ml-suite/overlaybins/aws/overlay_3.xclbin
AFI not yet loaded, proceed to download.
ERROR: Failed to create compute program from binary -44
```
Could someone help me out with this?
Sincerely thanks!

b04902036, 2018-07-25:
By the way, shouldn't we first build .xclbin file into .awsxclbin and create AFI first? I haven't seen this part in the provided tutorial.

wilderfield, 2018-07-25:
The overlays aka "xclbin" files which are stored in ml-suite/overlaybins/aws are in actuality *.awsxclbin files. I have already created the AFIs for our users. They should work fine if you are using an instance in us-east. We dropped the *.awsxclbin naming convention to make it easier to run all our code on any platform. Sorry about this confusion. Also, can you start from the FPGA Developer AMI for now? We need to update the ml-suite AMI to support the included xclbins. The included xclbins were built for Amazon's 2017.4 DSA shell. https://aws.amazon.com/marketplace/pp/B06VVYBLZZ

b04902036, 2018-07-26:
Really thanks for your help! I currently use an instance in us-west-2, that's probably why I got the error. Is it possible for you to support different region to load your AFIs ? Or I'll have to change my region to us-east to use your product?
Thank you !

wilderfield, 2018-07-26:
You are welcome! Thanks for your patience in trying out the ml-suite. Once
we bundle up the new ml-suite AMI, the region problem will be automatically
handled when you launch the AMI. Also, I urge you to do your work on a
fresh FPGA Developer AMI to start off with, because that AMI includes the
latest Amazon shell, which is compatible with the xclbins we've included in
the repo.

wilderfield, 2018-07-26:
> Uh, just to comfirm, so I don't have to deal with the region and just launch a fresh FPGA Developer AMI?
> My instance is in us-west-2, do I need to set a special aws configure to load your AFIs in us-east?
> Thank you for your patience and answers!

Just use us east for now. Is there a major reason you want to be in a different region? If so I can look into testing it tomorrow.