Xilinx/ml-suite #24: Yolo2 demo error "AssertionError: Theshold is not a scalar"

adki, 2018-08-05:
When I run "./run.sh aws e2e" on AWS F1, it finishes with error as follows.

....
--------------------------------------------------------------------------------
Processing layer 1 of 54
Layer Name:conv0 Type:Convolution
Inputs: ['data'], Outputs: ['conv0']
Quantizing conv input layer ... conv0
Traceback (most recent call last):
  File "yolo.py", line 86, in <module>
    quantizer.quantize()
  File "./xfdnn/tools/quantize/quantize.py", line 157, in quantize
  File "./xfdnn/tools/quantize/quantize_caffe.py", line 110, in executeCalibration
  File "./xfdnn/tools/quantize/quantize_caffe.py", line 225, in preProcess
  File "./xfdnn/tools/quantize/quantize_base.py", line 142, in QuantizeThresholdBlob
AssertionError: Theshold is not a scalar
(ml-suite) [root@AWS]

wilderfield, 2018-08-06:
Hi,
When I test on the same AMI, I do not have any errors with ./run.sh aws e2e

Can you check the amount of available disk space on your instance? I know storage space can be tight on that AMI, because there is a 35G /swapfile

Also, in case it helps you, here is the output of conda list on my instance: 
```
(ml-suite) [centos@ip-172-31-42-154 yolo]$ conda list      
                                                                                                                                                                                  
packages in environment at /home/centos/anaconda2/envs/ml-suite:                                                                                                                                                                           
                                                                                                                                                                                                                                       
Name                    Version                   Build  Channel                                                                                                                                                                           
absl-py                   0.2.2                      py_0    conda-forge                                                                                                                                                                     
aniso8601                 3.0.2                     <pip>                                                                                                                                                                                    
astor                     0.7.1                      py_0    conda-forge                                                                                                                                                                     
backports                 1.0                      py27_1    conda-forge                                                                                                                                                                     
backports.functools_lru_cache 1.5                        py_1    conda-forge                                                                                                                                                                 
backports.shutil_get_terminal_size 1.0.0                      py_3    conda-forge                                                                                                                                                            
backports.weakref         1.0.post1                py27_0    conda-forge                                                                                                                                                                     
backports_abc             0.5                      py27_0    conda-forge                                                                                                                                                                     
blas                      1.1                    openblas    conda-forge                                                                                                                                                                     
bleach                    1.5.0                    py27_0    conda-forge                                                                                                                                                                     
boost                     1.66.0                   py27_1    conda-forge                                                                                                                                                                     
boost-cpp                 1.66.0                        1    conda-forge                                                                                                                                                                     
boto3                     1.7.64                    <pip>                                                                                                                                                                                    
botocore                  1.10.64                   <pip>                                                                                                                                                                                    
bzip2                     1.0.6                h470a237_2    conda-forge                                                                                                                                                                     
ca-certificates           2018.4.16                     0    conda-forge                                                                                                                                                                     
caffe                     1.0             py27_blas_openblas_202  [blas_openblas]  conda-forge                                                                                                                                               
cairo                     1.14.10                       0    conda-forge                                                                                                                                 [... cut ...]
widgetsnbextension        3.3.0                     <pip>
x264                      20131218                      0    conda-forge
xorg-libxau               1.0.8                h470a237_5    conda-forge
xorg-libxdmcp             1.1.2                h470a237_6    conda-forge
xz                        5.2.3                         0    conda-forge
yaml                      0.1.7                         0    conda-forge
zlib                      1.2.11               h470a237_3    conda-forge
```

adki, 2018-08-07:
First of all, thank you for your help, but it still does not work and ends with the same error "AssertionError: Theshold is not a scalar". [I think, it should be "Threshold" not "Theshold".]

1. The EBS has been enlarged.  30Gbyte --> 100Gbyte
(ml-suite) [root@AWS] df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/xvda1      100G   89G   12G  89% /
devtmpfs         60G     0   60G   0% /dev
tmpfs            60G  4.0K   60G   1% /dev/shm
tmpfs            60G   18M   60G   1% /run
tmpfs            60G     0   60G   0% /sys/fs/cgroup
/dev/xvdb        30G   28M   29G   1% /home/centos/src/project_data
tmpfs            12G   12K   12G   1% /run/user/42
tmpfs            12G     0   12G   0% /run/user/1000
(ml-suite) [root@AWS] free
              total        used        free      shared  buff/cache   available
Mem:      125671008     1268592   101131304       17676    23271112   123483656
Swap:      36700156           0    36700156
(ml-suite) [root@AWS] 

2. Re-installed Anaconda and ml-suite VM, but there are many differences comparing to your list.
   such as two versions of 'tensorboard' and 'tensorflow" exist, which are 1.9 and 1.8.
tensorboard               1.9.0                    py27_0    conda-forge
tensorboard               1.8.0                     <pip>
tensorflow                1.8.0                     <pip>
tensorflow                1.9.0                    py27_0    conda-forge
   I presume that version 1.9.0 has been removed while ml-suite VM is created, but its related packages such as numpy still exists and these are for 1.9.0 not 1.8.0. This cause binary incompatibility.

here is mine.

# packages in environment at /home/centos/anaconda2/envs/ml-suite:
#
# Name                    Version                   Build  Channel
absl-py                   0.3.0                      py_0    conda-forge
astor                     0.7.1                      py_0    conda-forge
backports                 1.0                      py27_1    conda-forge
backports.functools_lru_cache 1.5                        py_1    conda-forge
backports.shutil_get_terminal_size 1.0.0                      py_3    conda-forge
backports.weakref         1.0.post1                py27_0    conda-forge
backports_abc             0.5                      py27_0    conda-forge
blas                      1.1                    openblas    conda-forge
bleach                    1.5.0                     <pip>
boost                     1.66.0                   py27_1    conda-forge
boost-cpp                 1.66.0                        1    conda-forge
bzip2                     1.0.6                h470a237_2    conda-forge
ca-certificates           2018.4.16                     0    conda-forge
caffe                     1.0             py27_blas_openblas_202  [blas_openblas]  conda-forge
cairo                     1.14.10              he56eebe_1    conda-forge
certifi                   2018.4.16                py27_0    conda-forge
cloudpickle               0.5.3                      py_0    conda-forge
configparser              3.5.0                     <pip>
cycler                    0.10.0                     py_1    conda-forge
dask-core                 0.18.2                     py_0    conda-forge
dbus                      1.13.0               h3a4f0e9_0    conda-forge
decorator                 4.3.0                      py_0    conda-forge
entrypoints               0.2.3                     <pip>
enum34                    1.1.6                    py27_1    conda-forge
expat                     2.2.5                hfc679d8_1    conda-forge
ffmpeg                    3.2.4                         3    conda-forge
fontconfig                2.13.0               hd36ec8e_5    conda-forge
freetype                  2.8.1                hfa320df_1    conda-forge
funcsigs                  1.0.2                      py_2    conda-forge
functools32               3.2.3.2                  py27_2    conda-forge
futures                   3.2.0                    py27_0    conda-forge
gast                      0.2.0                      py_0    conda-forge
gettext                   0.19.8.1                      0    conda-forge
gflags                    2.2.1                hfc679d8_0    conda-forge
giflib                    5.1.4                h470a237_0    conda-forge
glib                      2.55.0               h464dc38_2    conda-forge
glog                      0.3.5                hfc679d8_0    conda-forge
graphite2                 1.3.11               hfc679d8_0    conda-forge
[... 103 lines cut ...]
subprocess32              3.5.2                    py27_0    conda-forge
tensorboard               1.9.0                    py27_0    conda-forge
tensorboard               1.8.0                     <pip>
tensorflow                1.8.0                     <pip>
tensorflow                1.9.0                    py27_0    conda-forge
termcolor                 1.1.0                      py_2    conda-forge
terminado                 0.8.1                     <pip>
testpath                  0.3.1                     <pip>
theano                    1.0.2                    py27_0    conda-forge
tk                        8.6.8                         0    conda-forge
toolz                     0.9.0                      py_0    conda-forge
tornado                   5.1              py27h470a237_1    conda-forge
traceback2                1.4.0                    py27_0    conda-forge
[... cut ...]
wheel                     0.31.1                   py27_1    conda-forge
widgetsnbextension        3.4.0                     <pip>
x264                      20131218                      0    conda-forge
xorg-libxau               1.0.8                h470a237_6    conda-forge
xorg-libxdmcp             1.1.2                h470a237_7    conda-forge
xz                        5.2.4                h470a237_0    conda-forge
yaml                      0.1.7                         0    conda-forge
zlib                      1.2.11               h470a237_3    conda-forge

adki, 2018-08-08:
I tried Yolo2 on N. Virginia, but it also ends with 
```AssertionError: Theshold is not a scalar```.<br>

Are there anyone who tried YOLO2 with ml-suite on AWS Develop FPGA AMI, recently?
If anyone succeed in running it, it implies I had done something wrong.
Otherwise, there might be minor mismatch or bug.

By the way, the classification example did work on N. Virginia region, which did not at Oregon region.

Thanks.

wilderfield, 2018-08-08:
Hey,

It works for me :-/

I would suggest that you try out the object detection notebook as well. It is also YOLOv2, but created for the purpose of showing the basic flow. Whereas apps/yolo was created to be used for optimal performance.

Can you please make sure this file was successfully downloaded in your env:
models/yolov2/caffe/fp32/yolov2.caffemodel

```
It should be around 200MB

(ml-suite) [centos@ip-172-31-84-45 yolo]$ ls -lart ../../models/yolov2/caffe/fp32/yolov2.caffemodel
-rw-rw-r-- 1 centos centos 196735411 Jul 20 04:28 ../../models/yolov2/caffe/fp32/yolov2.caffemodel
```

The AFIs do need to be enhanced to support other regions. Haven't been able to get to it yet.

adki, 2018-08-08:
Hi,
I glad to hear from you that YOLO2 worked for you, but I did not get it done yet due to the same error ```AssertionError: Theshold is not a scalar```.
[run_log.txt](https://github.com/Xilinx/ml-suite/files/2269364/run_log.txt)

1. I checked the yolo2.caffemodel and it is around 200Mbyte as follows
<details><summary>Click to expand</summary>

```
[centos@AWS] ls -lart
total 192144
-rw-rw-r-- 1 centos centos      6454 Aug  8 03:24 yolo_deploy_608.prototxt
-rw-rw-r-- 1 centos centos      6454 Aug  8 03:24 yolo_deploy_224.prototxt
drwxrwxr-x 3 centos centos        17 Aug  8 03:24 ../
drwxrwxr-x 2 centos centos        92 Aug  8 03:24 ./
-rw-rw-r-- 1 centos centos 196735411 Aug  8 03:42 yolov2.caffemodel
```
</details><br>

2. It does not work for my AMI on N. Virginia. Here is how I run Yolo.
<details><summary>Click to expand</summary>

```
[centos@AWS] source aws-fpga/sdaccel_setup.s
....
[centos@AWS] sudo su
[root@ip-172-31-90-240 work.ml]# source ml-suite/overlaybins/setup.sh aws
.....
[root@ip-172-31-90-240 work.ml]# source ~centos/.bashrc
[root@AWS] source activate ml-suite
(ml-suite) [root@AWS] cd ml-suite/apps/yolo
(ml-suite) [root@AWS] ./run.sh aws e2e
.... details will be found in the attached file.
```
</details><br>

I'm still trying to figure out what was wrong, but I cannot trace beyond  yolo.py and xyolo.py.

seroyer, 2018-08-09:
I'm on a team with 3 people who have all tried it.  I do not get the `AssertionError: Theshold is not a scalar` error, but the other 2 do.  Haven't been able to figure out what is different.

wilderfield, 2018-08-09:
Can you all please post your version of numpy? 
```
conda activate ml-suite
python -c "import numpy; print (numpy.version.version)"
```

seroyer, 2018-08-09:
I have two environments working:
local container (no FPGA): 1.11.0
AWS: 1.14.5

2 failing environments are both 1.15.0

esberglu, 2018-08-09:
I had a failing environment with numpy 1.15.0. I downgraded to 1.14.5 and was able to get past this error.

wilderfield, 2018-08-10:
This is great news... I'm confused as to how you guys experienced having different numpy versions. I would have hoped that our instructions would lead you to all have the same version. 

It could be that some anaconda package got updated...

My numpy version is 1.14.5

Very soon we will release an updated ml-suite AMI where all the setup will be done for you. So... it should get better.

Thanks for your info!

Bryan

adki, 2018-08-10:
It is very good to know that **numpy** should be a specific version for ml-suite.
The numpy version was *1.15.0* and downgraded to **1.14.5**.
By the way it should be carried out in the **ml-suite** virtual environment.
<details><summary>Click to expand</summary>

```
[centos@AWS] conda activate ml-suite
(ml-suite) [centos@AWS] python -c "import numpy; print (numpy.version.version)"
1.15.0
(ml-suite) [centos@AWS] pip uninstall numpy==1.15.0
... ...
  Successfully uninstalled numpy-1.15.0
(ml-suite) [centos@AWS] pip install numpy==1.14.5
... ...
Successfully installed numpy-1.14.5
(ml-suite) [centos@AWS] python -c "import numpy; print (numpy.version.version)"
1.14.5
(ml-suite) [centos@AWS] 
```
</details><br>

Now I tried to run YOLO2 again with root permission (see 'TEMPORARY NOTE' of https://github.com/Xilinx/ml-suite)
And my details as follows.
 - region: 'us-east-1c' (N. Virginia).
 - AMI: AWS FPGA Developer AMI 1.4.0

<details><summary>Click to expand</summary>

```
(ml-suite) [root@AWS] ./run.sh aws e2e
Running with images: /home/centos/work.ml/ml-suite/xfdnn/tools/quantize/calibration_directory/13923040300_b4c8521b4d_z.jpg
... ...
Compiling weights from: /home/centos/work.ml/ml-suite/models/yolov2/caffe/fp32/yolov2.caffemodel
Writing weights to directory yolov2.caffemodel_data
SUCCESS
... ...
oimage = 7647574936_ffebfa2bea_z.jpg
Saving new image with bounding boxes drawn as out/7647574936_ffebfa2bea_z.jpg
INFO: Saving results as results.json
(ml-suite) [root@AWS]
```
</details><br>

The entire log file has been attached here. 
[yolo_log.txt](https://github.com/Xilinx/ml-suite/files/2276333/yolo_log.txt)

One of output file is also here.
[image]

It is very nice to see the result. Thank you Bryan.