Xilinx/ml-suite #26: Where is " xilinx trained models" for YOLOv2

adki, 2018-08-06:
According to "https://github.com/Xilinx/ml-suite/blob/master/apps/yolo/README.md",
the "xilinx trained models" should be downloaded before running yolo.
"2. Download the xilinx trained models from Xilinx.com, save as models at the root of this repo".

Where it is?

wilderfield, 2018-08-06:
Hi,

At the moment, the models are stored on github.com using git lfs

After you install git lfs, you can do `git lfs pull` to bring down all the large files.

They should appear in ml-suite/models/

Please let me know if this resolves your issues.

Thanks,
Bryan

adki, 2018-08-07:
Hi,

I got it.  Following files are found in 'ml-suite/models/yolov2/caffe/fp32'.

-rw-rw-r-- 1 centos centos      6454 Aug  7 13:35 yolo_deploy_224.prototxt
-rw-rw-r-- 1 centos centos      6454 Aug  7 13:35 yolo_deploy_608.prototxt
-rw-rw-r-- 1 centos centos 196735411 Aug  7 13:35 yolov2.caffemodel

Thanks.

closed 2018-08-07 by adki