Xilinx/ml-suite #27: "x264=20131218" causes error for creating conda ml-suite environment

adki, 2018-08-06:
When creating ml-suite environment of Anaconda2, '264=20131218' causes error.
Does it really necessary to install?
If so, what is correct one.

Thank you for this ML Suite.

wilderfield, 2018-08-06:
Hey,

It should be "x264". This worked for me a week or two ago. 

We are attempting to use Anaconda to bring in a prebuilt version of Caffe. That prebuilt version has a dependency on a certain version of the x264 library. x264=20131218 is specifying anaconda to bring in the correct version of the library. 

You should be able to copy and paste this command exactly.
conda create --name ml-suite python=2.7 x264=20131218 caffe pydot pydot-ng graphviz keras scikit-learn -c conda-forge

adki, 2018-08-07:
Hi,

I am not quite sure, but it installs x264 when I re-install (create) ml-suite VM after removing the environment using 'conda-env remove -n ml-suite'.

Thanks.

closed 2018-08-07 by adki