Xilinx/ml-suite #31: Can't clone ml-suite (repository is over its data quota)

cjvolzka, 2018-08-09:
Attempting to clone the repo to my AWS instance fails with the following:

```
[centos@ip-172-31-91-124 ~]$ git clone https://github.com/Xilinx/ml-suite.git
Cloning into 'ml-suite'...
remote: Counting objects: 36628, done.
remote: Compressing objects: 100% (322/322), done.
remote: Total 36628 (delta 144), reused 263 (delta 78), pack-reused 36210
Receiving objects: 100% (36628/36628), 743.40 MiB | 39.01 MiB/s, done.
Resolving deltas: 100% (13127/13127), done.
Downloading models/caffe/aiotlabs/fp32/cifar_normalized.npz (255 MB)
Error downloading object: models/caffe/aiotlabs/fp32/cifar_normalized.npz (6b97b94): Smudge error: Error downloading models/caffe/aiotlabs/fp32/cifar_normalized.npz (6b97b94a15c87f0b9143c3c567ca208a54ebdb3adc8cc0a9e66c392332fbe3b4): batch response: This repository is over its data quota. Purchase more data packs to restore access.

Errors logged to /home/centos/ml-suite/.git/lfs/logs/20180809T132055.292163089.log
Use `git lfs logs last` to view the log.
error: external filter git-lfs smudge -- %f failed 2
error: external filter git-lfs smudge -- %f failed
fatal: models/caffe/aiotlabs/fp32/cifar_normalized.npz: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'
```

I get something similar if I try `git lfs pull` afterward:
```
[root@ip-172-31-91-124 ml-suite]# git lfs pull
batch response: This repository is over its data quota. Purchase more data packs to restore access.                                                                                                                     
error: failed to fetch some objects from 'https://github.com/Xilinx/ml-suite.git/info/lfs'
```

kamranjk, 2018-08-09:
Hi,

Please try again. You should be able to download now. 

thanks,
kamran

cjvolzka, 2018-08-09:
It's letting me start the download now. Thanks Kamran!

closed 2018-08-09 by cjvolzka

adu81020799, 2018-09-21:
I am getting the same error. I am unable to download .Is there any work around this ?