Xilinx/ml-suite #33: Failed to allocate A device memory

esberglu, 2018-08-13:
We're trying to run the YOLO Object detection tutorial using our own YOLO model. With our custom caffemodel and prototxt we are able to get through compilation and quantization. But then we hit the following error trying to load the weights to the FPGA. Any ideas?

Loading weights/bias/quant_params to FPGA...
ERROR: Failed to allocate A device memory
python: xblas.cpp:3202: void xblasLoadA(XBLASHandle&, int, const void*, XBLASConfig*, int): Assertion `ret' failed.

wilderfield, 2018-08-13:
Usually this means either the weights directory was not created, or the path to the directory was not correctly passed to the load weights api.

esberglu, 2018-08-13:
I'm actually hitting this using the demo caffemodel as well, not just our custom model.

adki, 2018-08-14:
I did see the same error when I ran YOLO example. It happened when I ran YOLO twice and the second run ended with the error.
So, I stopped F1 instance and then stated F1. Then it runs.
There might be something missing to de-allocate/release resources or cleanup steps after run.
<details><summary>Click to expand</summary>

```
... ...
[XDNN] kernel configuration
[XDNN]   num cores       : 1
[XDNN]   dsp array width : 56
[XDNN]   img mem size    : 5 MB
[XDNN]   version         : 2.2
[XDNN]   8-bit mode      : 0
[XDNN]   Max Image W/H   : 1023
[XDNN]   Max Image Depth : 4095

Loading weights/bias/quant_params to FPGA...
ERROR: Failed to allocate A device memory
```
</details></br>

wilderfield, 2018-08-14:
Check your available disk space as well. If you are on the edge of a full hard disk this could happen.

esberglu, 2018-08-14:
I was able to run with the demo yolo model after restarting my F1 instance as well.

esberglu, 2018-08-14:
Some more information. After restarting my instance I am able to run the demo with the default model successfully multiple times in succession. But if I try to use my custom yolo model, I get the error no matter what (after a restart and after successful default model run). Once I hit the error using my custom model, any subsequent runs (default or custom) will also hit the error.

My guess is that there is an issue with the custom model that is causing this failure which puts the system in a bad state which is resolved by a restart. So really there are two issues here. 1) What is wrong with our model that is breaking the demo. 2) Why is the initial failure causing all subsequent runs to fail.

wilderfield, 2018-08-14:
Eric, would you be willing to share your model via Google Drive? I can try to reproduce. Just to confirm, does your model have reorg, or Leaky Relu? My email is shared in my github profile.

esberglu, 2018-08-14:
We had an issue with the caffemodel datadir having an extra directory in the path when using our custom model. The result was that the weights weren't being found. That's why we would see the issue initially. Still not sure why the initial failure causes all subsequent runs to fail.

cjvolzka, 2018-08-22:
I recently hit this as well experimenting on the yolo demo. I'm not sure if my situation is the same as others but running  ```fpga-clear-local-image -S 0``` cleared the loaded AFI image from the FPGA and fixed it without needing to restart the whole instance. (Note: I only have one FPGA in my instance and -S 0 clears the FPGA at the first slot. If you have multiple FPGAs, you might need to adjust the slot number)

closed 2018-09-07 by esberglu