Xilinx/ml-suite #34: Failure loading weights to FPGA

esberglu, 2018-08-14:
I've worked through https://github.com/Xilinx/ml-suite/issues/33 which was a misconfigured datadir for our custom yolo caffemodel. However, I'm still seeing an error trying to load the weights to the FPGA.

Loading weights/bias/quant_params to FPGA...
python: xdnn.cpp:4070: int XDNNFillWeightsBiasQuantBlob(short int*, int, std::string, std::string, const DType*, unsigned int, float, const DType*, unsigned int, float, short unsigned int, short unsigned int, unsigned int, unsigned int) [with DType = float; std::string = std::basic_string<char>]: Assertion `fpgaCfgFile.empty() || qp != __null' failed.

wilderfield, 2018-08-15:
> python: xdnn.cpp:4070: int XDNNFillWeightsBiasQuantBlob(short int*, int, std::string, std::string, const DType*, unsigned int, float, const DType*, unsigned int, float, short unsigned int, short unsigned int, unsigned int, unsigned int) [with DType = float; std::string = std::basic_string]: Assertion `fpgaCfgFile.empty() || qp != __null' failed.

I hit this exact error the day before you. (likely) The cause: if you ran some model, a weights directory is created, and text files containing weights are created. Say you change models, but you don’t change the name of that directory... it won’t purge the old folder, it will just overwrite some of the files. Long story short, remove the _data directory before you run your own model.

DavidBegert, 2018-12-15:
@wilderfield I am getting the same error when trying to run a custom segmentation model (even after removing the _data folder and restarting). Any other ideas? Thanks :)

wilderfield, 2019-01-30:
I’m closing this since the errored function mentioned does not exist in latest code base.

closed 2019-01-30 by wilderfield