Xilinx/ml-suite #35: Gemx to ml-suite

b04902036, 2018-08-15:
Thanks again for this great work on FPGA!
However, I have some question about ml-suite.
I see `gemx` in this repo, so I guess the xclbins are built from gemx?
But the spec in your `gemx` repo says that the gemx engine only support uint16, int16, short and unsigned short as its t_FloatType, so how do you achieve 8 bit (char) computation?????
Or the `gemx` in this repo is different from your gemx repo?
Thank you!

bill-teng, 2018-08-15:
Hi, 

The gemx in this repository is essentially a mirror of the separate gemx public repo. However, we purposely left out hardware details of gemx as this repo is taking more of a software-centric view of FPGA acceleration. 

We will add more software-centric documentation for gemx ( renamed to "xfmlp" for future releases) in the future.

As for adding 8-bit computation, software support for it will be fairly trivial, but will require some modifications. As for hardware changes, you should start the hardware conversation in the separate gemx repo.

Regards,

Bill

b04902036, 2018-08-15:
Got it, thanks!

closed 2018-08-15 by b04902036