Xilinx/ml-suite #43: Yolo script output freezes out:

adu81020799, 2018-09-24:
I am currently running Yolo demo on the FPGA AMI. 
I have followed the steps mentioned in the ml-suit setup for FPGA AMI.

    git clone https://github.com/aws/aws-fpga.git
    cd aws-fpga
    source sdaccel_setup.sh
 and then I run the ./run script. But the script freezes at the following point for which i need to give a keyboard interrupt to stop it. Following was the last screen . It just runs for single image.

Loading weights/bias/quant_params to FPGA...
Finished batch 2
INFO: Running Image(s):
INFO: ['/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/13923040300_b4c8521b4d_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/14931486720_37bd588ce9_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/15439525724_97d7cc2c81_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/16247716843_b419e8b111_z.jpg']
INFO: Preparing Input...
INFO: Running 4 image(s)
INFO: 
Total FPGA: 655.149937 ms
INFO: Image Time: (163.787484 ms/img):
INFO: Running Image(s):
Finished batch 3
INFO: ['/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/3272651417_27976a64b3_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/3591612840_33710806df_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/36085792773_b9a3d115a3_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/4788821373_441cd29c9f_z.jpg']
INFO: Preparing Input...
INFO: Results for image 0: /home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/13923040300_b4c8521b4d_z.jpg
INFO: Running 4 image(s)
INFO: 
Total FPGA: 290.031910 ms
INFO: Image Time: (72.507977 ms/img):
INFO: Running Image(s):
INFO: ['/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/4814953542_de4b973dc2_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/5904386289_924b24d75d_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/7291910830_86a8ebb15d_z.jpg', '/home/centos/ml-suite/xfdnn/tools/quantize/calibration_directory/7647574936_ffebfa2bea_z.jpg']
INFO: Preparing Input...
INFO: Running 4 image(s)
INFO: Found 2 boxes
INFO: Obj 0: skateboard
INFO: 	 score = 0.622609
INFO: 	 (xlo,ylo) = (277,381)
INFO: 	 (xhi,yhi) = (509,311)
INFO: Obj 1: person
INFO: 	 score = 0.865611
INFO: 	 (xlo,ylo) = (149,340)
INFO: 	 (xhi,yhi) = (504,28)
DEBUG: STREAM 'IHDR' 16 13
DEBUG: STREAM 'IDAT' 41 1216
oimage = 13923040300_b4c8521b4d_z.jpg
Saving new image with bounding boxes drawn as out/13923040300_b4c8521b4d_z.jpg
X11 connection rejected because of wrong authentication.
QXcbConnection: Could not connect to display localhost:10.0
INFO: 
Total FPGA: 290.164948 ms
INFO: Image Time: (72.541237 ms/img):
.

adu81020799, 2018-09-24:
Hello, 
Thank you for the quick update. I will try your solution. Currently i think there is a single picture which has got bounding box. I wanted to visualize this picture. Can you send me the SCP command to send the files back from the instance to my local machine ? I tired several ways like 
scp -i home/centos.Jhanani1.pem centos@ec2-54-173-231-1.compute-1.amazonaws.com: ml-suite/apps/yolo/out/13923040300_b4c8521b4d_z.jpg
but it didnt work. 
And thanks a lot for sending  me the models files.

regards

adu81020799, 2018-09-24:
hello,
Your solution worked. It now shows me an image, but shouldn't it be running for more images?Like 4 images ?
Regards
Adarsh

wilderfield, 2018-09-24:
The code is calling cv2.imshow and waiting for the user to hit a key, before it will display the next.

Drawing boxes, writing output images, and displaying those images is not something that is done for production. It’s just a debug feature. You can change it if you want a different behavior. The code is in xyolo.py

adu81020799, 2018-09-24:
Hello,
It did work. 
Thanks a lot

closed 2018-09-24 by adu81020799