Xilinx/ml-suite #44: Unable to open FPGA handle through the Jupyter notebooks

adu81020799, 2018-10-01:
I was trying to run the new developer notebook which was added recently from here:
https://github.com/Xilinx/ml-suite/blob/master/notebooks/Xilinx-ML-Developer-Lab/ml-suite-developer-lab.ipynb.
I launched the notebook using the new start_ami.sh file (this was really help full though kudos for you all). But in the last step i was not able to create the FPGA handle. I know that we need sudo permission to access the FPGA thus i tried to change the script to mask off the sudo users but then the jupyter failed to load notebooks. I notice the same error when I try to launch the jupyter notebook from the ml-suite environment.
Please provide your thoughts where I might be going wrong.
@wilderfield

kamranjk, 2018-10-09:
hi,

this lab will work better with the aws ml suite ami october update. It will be available shortly. 

thanks,
kamran

wilderfield, 2019-01-30:
https://aws.amazon.com/marketplace/pp/B077FM2JNS

Available as of dec.

After starting go to

PublicDNS:8888

In web browser

closed 2019-01-30 by wilderfield