Xilinx/ml-suite #46: failure for the darknet2caffe.py script for tinyDarknet

adu81020799, 2018-10-04:
Hi, I wanted to run the tiny dark net application using the ML suite. So for the initial step I downloaded the Weights  and CFG files from  here .https://pjreddie.com/darknet/tiny-darknet/. When I ran the script as
python darknet2caffe.py -d "tinydark/tiny.cfg" -w "tinydark/tiny.weights" -p"prottiny" -c"cafemodeltiny"
I got the following error:
Traceback (most recent call last):
  File "darknet2caffe.py", line 516, in <module>
    darknet2caffe(cfgfile, weightfile, protofile, caffemodel, arch, mergeBN)
  File "darknet2caffe.py", line 84, in darknet2caffe
    start = load_conv2caffe(buf, start, params[conv_layer_name])
  File "darknet2caffe.py", line 120, in load_conv2caffe
    conv_param[0].data[...] = np.reshape(buf[start:start+weight.size], weight.shape); start = start + weight.size
  File "/home/centos/anaconda2/envs/ml-suite/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 257, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/home/centos/anaconda2/envs/ml-suite/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 52, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
Has any one tried to compile protext and .caffe file for Tiny dark net.?

wilderfield, 2018-10-05:
Hey,

It has unfortunately only been tested with yolov2.cfg and tiny_yolov2.cfg

Would love your help to enhance it, if it is broken for tiny-darknet.

It could be there are some layers in tiny-darknet that aren't quite
supported by the converter script.

adu81020799, 2018-10-05:
Hello,
I tried today for tinyyolo_cfg from here https://github.com/pjreddie/darknet/blob/master/cfg/yolov2-tiny.cfg.
It still gave me error as
Traceback (most recent call last):
  File "darknet2caffe.py", line 528, in <module>
    darknet2caffe(cfgfile, weightfile, protofile, caffemodel, arch, mergeBN)
  File "darknet2caffe.py", line 83, in darknet2caffe
    start = load_conv_bn2caffe(buf, start, params[conv_layer_name], params[bn_layer_name], params[scale_layer_name])
  File "darknet2caffe.py", line 172, in load_conv_bn2caffe
    conv_param[0].data[...] = np.reshape(buf[start:start+conv_weight.size], conv_weight.shape); start = start + conv_weight.size
  File "/home/uncc_embed/anaconda2/envs/ml-suite/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 257, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/home/uncc_embed/anaconda2/envs/ml-suite/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 52, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 650343 into shape (512,256,3,3)
If you can run the script with tiny yolo can you send me the link to the weights files and the .cfg files ?

Regards
Adarsh

adu81020799, 2018-10-10:
I think the problem was in the script, the script was not allocating enough memory for the buf np.array and it was missing by single location.Thus i increased the size of the array by using the np.pad command . The files got generated but I need to test whether these files are working or not. The pretext and  .caffe models.

closed 2018-10-11 by adu81020799