Xilinx/ml-suite #49: Handling  the unsupported layer

adu81020799, 2018-10-11:
Hello, 
I am trying to compile the tiny dark net using the ML-suite. I could generate the caffe models using the darknettocaffe script. But when I tried to compile the network using the compiler_caffe I got the following message:
* GENERATING OUTPUT FILES
**************************************************
XDNN Command file: work/tinydarknet/tinyDarknet.cmds
XDNN JSON Report file: work/tinydarknet/tinyDarknet.cmds.json
OUTPUT REPORT:
Unsupported Layers: 1
0) data
	Attributes: ("# LAYER data [u'Input'] ['layer']", u'data: type=Input, sizes=None, shapes=None, sched 0 Kernel None Strides None Padding None  NO VALID CODE  ')
Compiling weights from: ../tinydarknet/tiny.caffemodel
Writing weights to directory work/tinydarknet/tiny.caffemodel_data
SUCCES. 
Looks like it cannot  support first layer. But as i have seen in the other examples the first layer is always not supported and we start with conv0 layer . 
Can i just ignore this error?

wilderfield, 2018-10-11:
This is not really an error or a warning...

It is almost meaningless. It is just saying that there is no hardware
instruction created for the data layer.

The runtime handles moving the input to the accelerator.

I've asked the compiler developer to remove the reporting of the data layer
as an unsupported layer. We will see what happens.

Bryan

adu81020799, 2018-10-11:
Okay . Thank you.

closed 2018-10-11 by adu81020799