Xilinx/ml-suite #51: Running on FPGA Developer AMI

wilderfield, 2018-10-13:
Xilinx is still rolling out an updated AWS marketplace AMI for ml-suite. 

The new AMI will help developers begin quickly evaluating ml-suite without having to install anaconda, and prepare other dependencies.

Until then, you will need to evaluate with the FPGA Developer AMI. Please consider the below notes.

TEMPORARY NOTE:
If you are evaluating on AWS, the binaries we have included (overlaybins/aws/) support the latest Amazon Shell
DSA name: xilinx_aws-vu9p-f1-04261818_dynamic_5_0
The Xilinx ml-suite AMI was bundled for an older shell
For this reason, if you are starting your evaluation today, it is best to begin from the FPGA Developer AMI:
If you are using the AWS EC2 F1 FPGA DEVELOPER AMI the following steps are necessary to setup the drivers:

git clone https://github.com/aws/aws-fpga.git
cd aws-fpga
source sdaccel_setup.sh
Remember that AWS requires users to run as root to control the FPGA, so the following is necessary to use Anaconda as root:

Become root sudo su
Set Environment Variables Required by runtime source <MLSUITE_ROOT>/overlaybins/setup.sh aws
Set User Environment Variables Required to run Anaconda source ~centos/.bashrc
Activate the users Anaconda Virtual Environment  = source activate ml-suite
You can avoid disk space problems on the FPGA DEVELOPER AMI by creating an instance with more than the default 70G of storage, or by resizing the /swapfile to something less than 35G.

ZaharAd, 2018-11-25:
Hello, 
I can not complete the process because of this failure -

`
[root@ip-xxx-xx-xx-xxx-ml-suite]# source overlaybins/setup.sh aws
bash: /opt/xilinx/xrt/setup.sh: No such file or directory
20174
make: Entering directory `/artsim/workspace/users/zahara/ml-suite/apps/yolo/nms'
cd ./nms_20180209 && make
make[1]: Entering directory `/artsim/workspace/users/zahara/ml-suite/apps/yolo/nms/nms_20180209'
make[1]: Nothing to be done for `all'.
make[1]: Leaving directory `/artsim/workspace/users/zahara/ml-suite/apps/yolo/nms/nms_20180209'
make: Leaving directory `/artsim/workspace/users/zahara/ml-suite/apps/yolo/nms'
`

Its happening when I try to set the environment variables required by runtime
Does anyone know how to deal with this?

wilderfield, 2018-11-25:
There is no error in your log.

The script is trying to find a file on disk to determine the runtime
version. It doesn’t find it and defaults to the 2017.4 version.