Xilinx/ml-suite #61: loading AFI error

ZaharAd, 2018-11-26:
I trying to run Image Classification with Python APIs -
https://github.com/Xilinx/ml-suite/blob/master/examples/classification/README.md

In EU (Ireland) region, and got an error while loading the AFI

```
(ml-suite) [root@ip-xxx-xx-xx-xxx classification]# ./run.sh -p aws -t test_classify -k med -b 16
/ml-suite/overlaybins/setup.sh: line 20: /opt/xilinx/xrt/setup.sh: No such file or directory
20174
make: Entering directory `/ml-suite/apps/yolo/nms'
cd ./nms_20180209 && make
make[1]: Entering directory `/ml-suite/apps/yolo/nms/nms_20180209'
make[1]: Nothing to be done for `all'.
make[1]: Leaving directory `/ml-suite/apps/yolo/nms/nms_20180209'
make: Leaving directory `/ml-suite/apps/yolo/nms'

Running:
 Test: test_classify
 Model: googlenet_v1
 Fpgaoutsz: 1024
 Platform: aws
 Xclbin: overlay_1.xclbin
 Kernel Config: med
 Precision: 16
 Accelerator: 0

[XBLAS] # kernels: 1
[0]user:0xf010:0x1d51:[xocl:2017.4.5:128]
xclProbe found 1 FPGA slots with xocl driver running
Linux:3.10.0-693.21.1.el7.x86_64:#1 SMP Wed Mar 7 19:03:37 UTC 2018:x86_64
Distribution: CentOS Linux release 7.5.1804 (Core) 
GLIBC: 2.17
--- 
XILINX_OPENCL="/ml-suite/overlaybins/aws"
LD_LIBRARY_PATH="/ml-suite/overlaybins/aws/runtime/lib/x86_64/:/ml-suite/xfdnn/rt/xdnn_cpp/build/lib:/ml-suite/xfdnn/rt/lib:/ml-suite/ext/boost/lib:/ml-suite/ext/zmq/libs:/ml-suite/examples/classification"
--- 
CL_PLATFORM_VENDOR Xilinx
CL_PLATFORM_NAME Xilinx
CL_DEVICE_0: 0x1ff3b40
CL_DEVICES_FOUND 1, using 0
loading /ml-suite/overlaybins/aws/overlay_1.xclbin
AFI not yet loaded, proceed to download.
ERROR: Failed to create compute program from binary -44
```

I noticed this issue - 
https://github.com/Xilinx/ml-suite/issues/22

Written in July and have the same problem, there recommends creating a new f1 instance in us-east
is this steel the issue? I prefer not to mix regions because of the many instances that run for me in Ireland

wilderfield, 2018-11-26:
I believe I’ll need to copy the Fpga image to the EU region for you to use
it there. Please give me 24 hours.

ZaharAd, 2018-11-27:
Please let me know when this happens

wilderfield, 2018-11-28:
@ZaharAd we released our new ml-suite AMI yesterday.

You should have no problem launching that in the EU regions. Can you try it out, and let us know if this solves your problems?

https://aws.amazon.com/marketplace/pp/B077FM2JNS

ZaharAd, 2018-11-29:
Thanks! the problem is solved

closed 2018-12-02 by ZaharAd