Xilinx/ml-suite #62: Low Performance for Yolo V2 while running detection on ML Suite

MarouanGit, 2018-12-06:
Hi, 

This is more an observation rather than issue and a call for possible improvements. I'm running the ML-Suite and YoloV2 retrained on an Amazon Instance F1 running the Xilinx-FPGA template. I'm obtaining an average inference of 80ms/image compared to 20ms/image on a GPU. 

My goal is to understand two main things : 
- Is it a normal and expected inference time on the Alveo200 ? 
- If not what kind of improvements could be done (loading images seperatly in fpga memory for example ?) 

Attached the output of the detection on a F1 instance.  Any help would be appreciated !  

Many thanks, Cheers. 

[shell_results_xilinx.txt](https://github.com/Xilinx/ml-suite/files/2654388/shell_results_xilinx.txt)

wilderfield, 2019-01-30:
Your latency observations are correct for yolov2 on XDNNv2. 

Yolo typically takes a 608x608 input, and hence incurs large activations, and a heavy data movement penalties. The v2 architecture stalls the systolic array while partial activations are moved off chip.

XDNNv3 will alleviate this, and we should improve to around 25-30 ms. 

Stay tuned.

closed 2019-01-30 by wilderfield

wilderfield, 2019-01-30:
As per accuracy. The yolov2 model we have when ran on FPGA achieved 15.4% mAP as opposed to the published 21.4%. This on ms coco. We did not train our model very long, and we remove leaky relu.

MarouanGit, 2020-07-17:
Hi, 

I reopen this thread in order to see if the XDNNv3 has been released and if you had the opportunity to test YoloV3 on it ? How is the inference on this new architecture ?