Xilinx/ml-suite #68: undefined symbol: _ZN7leveldb2DB4OpenERKNS_7OptionsERKSsPPS0_

eerobert, 2019-01-28:
Clean install v1.3
And after create ml-suite conda envrionment, when running python -c "import caffe" it reports the above error. 
Any comments?

anjn, 2019-01-28:
Try this. It seems the dependency is broken inside Anaconda now.

```
conda activate ml-suite
conda install --no-deps leveldb=1.20=h00d4201_1 -c conda-forge
conda install --no-deps glog=0.3.5=hfc679d8_1 -c conda-forge
```

wilderfield, 2019-01-28:
Yes, someone in the Anaconda world broke us. We will have to work on a solid work around. 

Another used did this:

conda install -c defaults protobuf libprotobuf

eerobert, 2019-01-28:
Thanks anjn, it works

wilderfield, 2019-01-30:
I’ve reopened this in hopes that new users see this sooner vs later.