Xilinx/ml-suite #72: image_classification_tensorflow example - Error in deploying FPGA model (xdnn.createhandle)

saurabh1993, 2019-02-04:
Hi,  I am able to compile and quantize the python tensorflow model but step 5 ,deployment fails. I am using aws ubuntu F1 instance. Is it possible to run the ml-suite without FPGA developer AMI?

>>> import xfdnn.rt.xdnn as xdnn
>>> ret, handles = xdnn.createHandle('../overlaybins/aws/overlay_3.xclbin')
[XBLAS] # kernels: 1
[0]user:0x1042:0x7:[???:??:0]
xclProbe found 1 FPGA slots with xocl driver running
ERROR: Load image failed.
ERROR: Sleep until load failed.
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
ERROR AwsXcl: PCI kernel bar attach failed for slot# 0
WARNING: AwsXcl isGood: kernel, global & mgmt bar are: ffffffff, ffffffff, ffffffff
WARNING: xclOpen Handle check failed
[0]user:0x1042:0x7:[???:??:0]
device[0].user_instance : 0
ERROR: Load image failed.
ERROR: Sleep until load failed.
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
ERROR AwsXcl: PCI kernel bar attach failed for slot# 0
WARNING: AwsXcl isGood: kernel, global & mgmt bar are: ffffffff, ffffffff, ffffffff
ERROR: xclOpen Handle check failed
ERROR: Failed to find an OpenCL platform

wilderfield, 2019-02-08:
Hey, you can use the ml-suite AMI now. It should help you jump past a lot of set up issues. After you start, do conda activate ml-suite

saurabh1993, 2019-02-08:
Hi, I used ml-suite AMI and it worked for google-net, but when I try to run inception_v3 model. It gives an error in line 
fpgaRT.execute(batch_array, fpgaOutput) stating that kernel height mismatch. Has xilinx tested this on inception_v3. If yes, can I get the inception_v3 pb file?