Xilinx/ml-suite #74: no such file xfdnn_compiler_caffe.py or quantize.py

Haffon, 2019-02-21:
the command line tool demo (xfdnn_compiler_caffe_vgg16.sh) in examples directory unable to run as only .pyc is provided under xfdnn directory. Fixed replace calling xfdnn_compiler_caffe.py with xfdnn_compiler_caffe.pyc.

Saums, 2019-02-22:
@Haffon, you changed the name to solve the problem? I'm facing this issue with a lot of files as only pyc files have been uploaded and no py.

Haffon, 2019-02-22:
@Saums Yes, I solved it by change the ".py" to ".pyc" in shell command files, it works. As in 
 ml-suite/examples/compile/xfdnn_compiler_caffe_places365.sh
Line 26 python $MLSUITE_ROOT/xfdnn/tools/compile/bin/xfdnn_compiler_caffe.pyc \
Line 35 python $MLSUITE_ROOT/xfdnn/tools/compile/bin/xfdnn_compiler_caffe.pyc \
Thanks.

mahinlma, 2019-06-01:
@Saums , try this also as alternative
pip install uncompyle6
uncompyle6 file.pyc >> file.py