Xilinx/ml-suite #75: models/caffe/deephi/fp32/inception_v1_pruned_v3_deploy.prototxt is missed in models_2018-12-14.zip

Haffon, 2019-02-21:
ml-suite/examples/compile/xfdnn_compiler_caffe_deephi.sh need models/caffe/deephi/fp32/inception_v1_pruned_v3_deploy.prototxt but unable to find it inside models_2018-12-14.zip with this download url: https://www.xilinx.com/products/acceleration-solutions/xilinx-machine-learning-suite.html#gettingStartedU200

wilderfield, 2019-02-21:
Hey,

The models that we acquired from the Deephi
acquisition are not being publicly released as of yet.

They are pruned versions of standard models.

Can you share the reason for your interest?

Perhaps I can help raise priority.

Bryan

Haffon, 2019-02-21:
I'm testing ml-suite with the help of avnet. 
thanks.