Xilinx/ml-suite #76: Raspi + Spartan 6

dadrake3, 2019-02-21:
I am looking to implement an edge ML inferencing accelerator for the raspi. I am wondering if it is possible to use leflow to compile a TensorFlow graph to run on the spartan 6 FPGA. I am not sure where to start and I am wondering if this would be possible within ml-suite

Haffon, 2019-02-22:
Hi, @dadrake3 we downloaded and installed xrt and xdma from https://www.xilinx.com/products/acceleration-solutions/xilinx-machine-learning-suite.html#gettingStartedU200 for cloud on ubuntu 16.04, you can check https://www.xilinx.com/products/design-tools/ai-inference/ai-developer-hub.html#edge for Xilinx edge AI solution.

wilderfield, 2019-02-22:
Hey, while leflow looks super cool, and interesting...

ml-suite is software that wraps a DNN engine that is relatively fixed.
Synthesis is already done. The tools are provided to map data flow graphs
to the engine. The engine is designed for 16nm FPGAs.

In the future we may add kernels that are optimized for specific networks.

dadrake3, 2019-02-22:
Im still a little confused, does the ml-suite not work with the spartan 6 fpga made by xilinx?

wilderfield, 2019-02-22:
Correct.

The hardware design coupled with ml-suite makes heavy use of a new silicon
feature called “URAM”. URAM cells are large on chip SRAM buffers that hold
network activations.

The kernels also use DDR4 controllers which don’t exist in the era of
Spartan 6.

Haffon, 2019-02-23:
@dadrake3 wilderfield is correct, xilinx only provide precompiled overlaybins for VCU1525, Alveo-U200(KU9P)/250(KU13P) and so on. Write you own DNN layer/kernel for spartan 6 may be interesting, How complex edge ML could be?

closed 2019-06-28 by wilderfield