Xilinx/ml-suite #77: Tensorflow functions to call

Saums, 2019-02-26:
I can see the files in the folders that are for compiling and quantizing tensorflow but not able to use them because I can't see the functions inside them. Any documentation support available?

If not please tell me how to solve the error below.

```
from __future__ import print_function
import os,sys
 
# Bring in Xilinx ML-Suite Compiler

from xfdnn.tools.compile.bin.xfdnn_compiler_tensorflow import TensorflowFrontend as xfdnnCompiler
```
Error:
```
 File "/home/saumya/ml-suite/xfdnn/tools/compile/bin/xfdnn_compiler_tensorflow.py", line 1
    �
      ^
SyntaxError: Non-ASCII character '\xf3' in file /home/saumya/ml-suite/xfdnn/tools/compile/bin/xfdnn_compiler_tensorflow.py on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
```
As I can't see the file because .pyc files are provided and not py.

wilderfield, 2019-02-27:
Can you try using this fix in the notebook?

https://github.com/cheind/py-motmetrics/issues/2

For now the python code for compiler & quantizer is not being delivered.

Bryan

Saums, 2019-03-02:
@wilderfield I tried that fix but it didn't fix the error. 

In case you personally have access to the python code, does the TensorflowFrontend function exist in that file? I took hint from the Caffe file and assumed function might be called that.

Saums, 2019-03-02:
I checked the notebook labelled 'image_classification_tensorflow' to understand the tensorflow related information. I will close the issue. Thank you!

closed 2019-03-02 by Saums