Xilinx/ml-suite #80: I want to use my pre-trained model using ml-suite but cannot run..

nediver, 2019-03-29:
I use my pre-trained model named INCEPTIONV3 but it cannot be operated.

When using ml-suite/notebooks/image_classification_tensorflow.ipynb, in Step 3. Define an xfdnnCompiler instance and pass it arguments, following error occurred.

It seems like DDR capacity problem, but I don't know how to solve this problem.
Is it anyone who use self pre-trained model using ml-suite?

Thank you. (Following is fail.log with my pre-trained model and pass.log with default model)

When using my pre-trained model
******************************************************************************************************
Reset Memory
Trying strategy bysize NO DDR:
Failed Allocation MemoryAllocation(start=0, end=3612672.0, size=3612672.0, extra=[], strategy=[], layout=-1, timestamp=-1, slice=0, shapes=SizeType(batches=1, channels=64, height=147, width=147), replication=Replication(full_sect_num=0, repl_sect_num=0, repl_unit_num=0, repl_unit_width=0, channels_division=0), written=False)
WARNING: NO SPACEAM for pinky
FREE SPACE 2 [4555776, 5242880, 687104]
	[0, 2749440, 2749440]
ALIVE BLOBS 1 InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[2749440,4555776] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=147, width=147)
PAST  BLOBS 2 input_blob M[0,918528] Z=918528 F=[1] B=[0] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=3, height=299, width=299)
	InceptionV3/InceptionV3/Conv2d_1a_3x3/Conv2D_blob M[918528,2749440] Z=1830912 F=[2] B=[1] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=149, width=149)
current  Bs InceptionV3/InceptionV3/Conv2d_2b_3x3/Conv2D_blob M[0,3612672] Z=3612672 F=[4] B=[3] E=[] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=64, height=147, width=147)
	InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[0,1806336] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=32, height=147, width=147)
Reset Memory
Trying strategy bottom NO DDR:
Failed Allocation MemoryAllocation(start=0, end=3612672.0, size=3612672.0, extra=[], strategy=[], layout=-1, timestamp=-1, slice=0, shapes=SizeType(batches=1, channels=64, height=147, width=147), replication=Replication(full_sect_num=0, repl_sect_num=0, repl_unit_num=0, repl_unit_width=0, channels_division=0), written=False)
WARNING: NO SPACEAM for pinky
FREE SPACE 2 [4555776, 5242880, 687104]
	[0, 2749440, 2749440]
ALIVE BLOBS 1 InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[2749440,4555776] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=147, width=147)
PAST  BLOBS 2 input_blob M[0,918528] Z=918528 F=[1] B=[0] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=3, height=299, width=299)
	InceptionV3/InceptionV3/Conv2d_1a_3x3/Conv2D_blob M[918528,2749440] Z=1830912 F=[2] B=[1] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=149, width=149)
current  Bs InceptionV3/InceptionV3/Conv2d_2b_3x3/Conv2D_blob M[0,3612672] Z=3612672 F=[4] B=[3] E=[] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=64, height=147, width=147)
	InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[0,1806336] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=32, height=147, width=147)
Reset Memory
Trying strategy top NO DDR:
Failed Allocation MemoryAllocation(start=0, end=3612672.0, size=3612672.0, extra=[], strategy=[], layout=-1, timestamp=-1, slice=0, shapes=SizeType(batches=1, channels=64, height=147, width=147), replication=Replication(full_sect_num=0, repl_sect_num=0, repl_unit_num=0, repl_unit_width=0, channels_division=0), written=False)
WARNING: NO SPACEAM for pinky
FREE SPACE 2 [0, 687104, 687104]
	[2493440, 5242880, 2749440]
ALIVE BLOBS 1 InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[687104,2493440] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=147, width=147)
PAST  BLOBS 2 input_blob M[4324352,5242880] Z=918528 F=[1] B=[0] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=3, height=299, width=299)
	InceptionV3/InceptionV3/Conv2d_1a_3x3/Conv2D_blob M[2493440,4324352] Z=1830912 F=[2] B=[1] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=149, width=149)
current  Bs InceptionV3/InceptionV3/Conv2d_2b_3x3/Conv2D_blob M[0,3612672] Z=3612672 F=[4] B=[3] E=[] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=64, height=147, width=147)
	InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[0,1806336] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=32, height=147, width=147)
Reset Memory
Trying strategy tops NO DDR:
Failed Allocation MemoryAllocation(start=0, end=3612672.0, size=3612672.0, extra=[], strategy=[], layout=-1, timestamp=-1, slice=0, shapes=SizeType(batches=1, channels=64, height=147, width=147), replication=Replication(full_sect_num=0, repl_sect_num=0, repl_unit_num=0, repl_unit_width=0, channels_division=0), written=False)
WARNING: NO SPACEAM for pinky
FREE SPACE 2 [0, 687104, 687104]
	[2493440, 5242880, 2749440]
ALIVE BLOBS 1 InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[687104,2493440] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=147, width=147)
PAST  BLOBS 2 input_blob M[4324352,5242880] Z=918528 F=[1] B=[0] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=3, height=299, width=299)
	InceptionV3/InceptionV3/Conv2d_1a_3x3/Conv2D_blob M[2493440,4324352] Z=1830912 F=[2] B=[1] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=149, width=149)
current  Bs InceptionV3/InceptionV3/Conv2d_2b_3x3/Conv2D_blob M[0,3612672] Z=3612672 F=[4] B=[3] E=[] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=64, height=147, width=147)
	InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[0,1806336] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=32, height=147, width=147)
Reset Memory
Trying strategy bottle NO DDR:
Failed Allocation MemoryAllocation(start=0, end=3612672.0, size=3612672.0, extra=[], strategy=[], layout=-1, timestamp=-1, slice=0, shapes=SizeType(batches=1, channels=64, height=147, width=147), replication=Replication(full_sect_num=0, repl_sect_num=0, repl_unit_num=0, repl_unit_width=0, channels_division=0), written=False)
WARNING: NO SPACEAM for pinky
FREE SPACE 2 [4555776, 5242880, 687104]
	[0, 2749440, 2749440]
ALIVE BLOBS 1 InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[2749440,4555776] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=147, width=147)
PAST  BLOBS 2 input_blob M[0,918528] Z=918528 F=[1] B=[0] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=3, height=299, width=299)
	InceptionV3/InceptionV3/Conv2d_1a_3x3/Conv2D_blob M[918528,2749440] Z=1830912 F=[2] B=[1] E=[1] S=['layer'] [] L=0 T=SizeType(batches=1, channels=32, height=149, width=149)
current  Bs InceptionV3/InceptionV3/Conv2d_2b_3x3/Conv2D_blob M[0,3612672] Z=3612672 F=[4] B=[3] E=[] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=64, height=147, width=147)
	InceptionV3/InceptionV3/Conv2d_2a_3x3/Conv2D_blob M[0,1806336] Z=1806336 F=[3] B=[2] E=[1] S=['layer'] [] L=-1 T=SizeType(batches=1, channels=32, height=147, width=147)
Reset Memory

Trying strategy bottles NO DDR:
Failed Allocation MemoryAllocation(start=0, end=3612672.0, size=3612672.0, extra=[], strategy=[], layout=-1, timestamp=-1, slice=0, shapes=SizeType(batches=1, channels=64, height=147, width=147), replication=Replication(full_sect_num=0, repl_sect_num=0, repl_unit_num=0, repl_unit_width=0, channels_division=0), written=False)
[... 43 lines cut ...]
Performing two level schedule strategy all
Done schedule 127 STEPS

~~~

Compilation Successful!

Network Operations Count: 11548368530
Failed to complete compilation: Unknown format code 'd' for object of type 'float'
******************************************************************************************************

When using default googlenet by xilinx
******************************************************************************************************
Reset Memory
Trying strategy bysize NO DDR:
Done Strategy bysize NO DDR
Done schedule 84 STEPS

~~~

Compilation Successful!

Network Operations Count: 3286771040
DDR Transfers (bytes): 0
******************************************************************************************************

wilderfield, 2019-03-29:
Hi,

This looks like a compiler bug. It seems like it is trying to print
something, that something is of type float, but it was specified as an
integer.

It looks like it actually succeeded compilation, but hit this relatively
minor snag.

Regardless, I will advise you to hold off on Inception V3. We have some
flow problems with Inception V3 that we are debugging internally. They will
likely be resolved in a few more weeks.

Bryan

nediver, 2019-04-01:
Hi, Thank you for your reply.
I have another question about ML-Suite.

I want to use my pre-trained Densenet model operating in ML-Suite system, actually.
But Densenet model prints same error with Inception V3 when I compile the model by jupyter notebook.
Is it same bug with Inception V3 and Densenet?
If it is same bug, when this bug is fixed?

Thank you very much for read my question.

nediver, 2019-04-10:
> Hi, This looks like a compiler bug. It seems like it is trying to print something, that something is of type float, but it was specified as an integer. It looks like it actually succeeded [...]

Hi, Thank you for your reply.
I have another question about ML-Suite.

I want to use my pre-trained Densenet model operating in ML-Suite system, actually.
But Densenet model prints same error with Inception V3 when I compile the model by jupyter notebook.
Is it same bug with Inception V3 and Densenet?
If it is same bug, when this bug is fixed?

Thank you very much for read my question.

wilderfield, 2019-06-28:
ML Suite 1.5 is the planned release to more broadly support Tensorflow. The previous support was really more of a proof of concept.

Mahdi89, 2019-08-04:
Any thoughts on when the new release is coming out?