Xilinx/ml-suite #81: Image classification using tensorflow but accuracy is not good. (googlenet, resnet50)

nediver, 2019-04-09:
I have question about using image_classification_tensorflow.ipynb using Alveo-U200.

When I test classification with image_classification_caffe.ipynb using googlenet, the example picture like pomeranian is classified like this.

---------- Prediction 1/1 for /home/et/ml-suite/examples/classification/dog.jpg ----------
0.7053 "n02112018 Pomeranian"
0.0794 "n02123394 Persian cat"
0.0450 "n02085620 Chihuahua"
0.0180 "n02094433 Yorkshire terrier"
0.0154 "n04399382 teddy, teddy bear"

but I test classification with image_classification_tensorflow.ipynb, the result is followed.
It cannot classification the pomeranian.

---------- Prediction 1/1 for /home/et/ml-suite/examples/classification/dog.jpg ----------
0.6001 "n02123394 Persian cat"
0.2714 "n02112018 Pomeranian"
0.0226 "n02086079 Pekinese, Pekingese, Peke"
0.0130 "n02085936 Maltese dog, Maltese terrier, Maltese"
0.0045 "n02112137 chow, chow chow"

When I test classification with image_classification_tensorflow.ipynb using resnet50, the result is not correct also.

---------- Prediction 1/1 for /home/et/ml-suite/examples/classification/dog.jpg ----------
0.5223 "n03388043 fountain"
0.2058 "n04399382 teddy, teddy bear"
0.0669 "n03124170 cowboy hat, ten-gallon hat"
0.0457 "n02807133 bathing cap, swimming cap"
0.0308 "n04589890 window screen"

I use default code but the result is correct when using caffe with googlenet only.
What is the problem with the result?

wilderfield, 2019-04-09:
Hey,

Even though it’s the same network architecture, the caffe model, and tensor
flow model are not the same. They were trained in different frameworks, and
I’m not sure that each were trained to the same quality.

To verify, you can run both models in cpu mode, and compare the output to
the FPGA output.

You are right that ResNet is not giving the correct predictions. The
current quantizer is insufficient for ResNets. There was a hack in place to
make it work, but the hack doesn’t work in 1.3.

When we run resnet in test classify, we run a prequantized version produced
using an internal tool.

In a week or two we'll have an update with better quantizer.

nediver, 2019-04-09:
Thank you for your answer.

You commented followed.

To verify, you can run both models in cpu mode, and compare the output to
the FPGA output.

And how can I use cpu mode in ml-suite?

nediver, 2019-04-10:
> Hey, Even though it’s the same network architecture, the caffe model, and tensor flow model are not the same. They were trained in different frameworks, and I’m not sure that each were trained to [...]

Thank you for your answer.

You commented followed.

To verify, you can run both models in cpu mode, and compare the output to
the FPGA output.

And how can I use cpu mode in ml-suite?