Xilinx/ml-suite #92: Link not found

asim-xlnx, 2019-06-17:
On the faq(https://github.com/Xilinx/ml-suite/blob/master/docs/faq.md) page the below mentioned links are not working:

Compile the model
a. See compiler tutorial (https://github.com/Xilinx/ml-suite/blob/notebooks/compiler_caffe.ipynb)
Quantize the model
a. See quantizer tutorial (https://github.com/Xilinx/ml-suite/blob/notebooks/quantizer_caffe.ipynb)

durgabhavaniv, 2019-06-26:
Hi @asim-xlnx 

For running image classification on terminal for caffe model, You can go through the below link.
https://github.com/durgabhavaniv/ml-suite/blob/master/docs/image_classifier_terminal_caffe.md

Please comment if you are facing some issues.

wilderfield, 2019-06-28:
@asim-xlnx , thanks for calling this out. We did get rid of the compiler and quantizer notebooks, because the image_classification_caffe.ipynb shows those two steps, and we didn't see the value in having them be seperate tutorials. It was creating too much extra work to keep them current, for no real extra benefit.

closed 2019-06-28 by wilderfield

durgabhavaniv, 2019-07-01:
Hi @wilderfield ,

image_classification_caffe.ipynb have some issue. So ruuning it on terminal make easy to debug.
Caffe_run.py added which make runtime easier.
view it on: https://github.com/durgabhavaniv/ml-suite/blob/master/examples/caffe/caffe_run.py

gguasti, 2019-07-17:
the link is missing at the bottom of the page

https://github.com/Xilinx/ml-suite/blob/master/docs/ml-suite-overview.md

<<With xfDNN you can connect to other Open Source frameworks and software easily with our Python APIs. For more information on this, please take a look at the xfDNN tool notebooks **here**>>