Xilinx/ml-suite #93: ml-suite 1.4 still not working on caffe image classification

eerobert, 2019-06-28:
Hi, 

i cannot run thru the example of caffe image classification
it complains about missing output files after quantize 
all i have is the quantize_train_test.prototxt but not deploy.prototxt and deploy.caffemode

wilderfield, 2019-06-28:
Lets continue this on [the forum.](https://forums.xilinx.com/t5/Xilinx-ML-Suite/bd-p/ML)
Please Confirm:
1) You are running inside of Docker
2) You have verified the canned examples work inside of examples/caffe
3) You are using a model_train_val.prototxt with pointers to your calibration data set

What model are you running? Is it open source?

Thanks,
Bryan

closed 2019-06-28 by wilderfield

eerobert, 2019-06-28:
i was following the setup steps from "install xrt" and "start docker container" as well as "jupyter notebooks setup"

yes. it is run insider docker. otherwise i cannot run those setup script

but the canned examples caffe image classification not working at all.

when i run to quantize steps, it seems like the quantize only output quantize_train_test.prototxt but not all 5 output files as describe in the readme. and hence cannot step forward

durgabhavaniv, 2019-07-01:
Hi @eerobert,

For running image classification on terminal for caffe model, You can go through the below link.
https://github.com/durgabhavaniv/ml-suite/blob/master/docs/image_classifier_terminal_caffe.md

Please comment if you are facing some issues.

eerobert, 2019-07-02:
Hi, 

Illegal instruction (core dumped) at decent_q
only quantize_train_test.prototxt at quantize_results folder

Just wonder why need to fork the github from original Xilinx one?

durgabhavaniv, 2019-07-02:
Hi @eerobert,

I have given pull request for this image classification on terminal.

Will you please elaborate at which step you are facing issue?

wilderfield, 2019-07-02:
Hey,

When we build caffe_decent we are building on a machine that is likely more
modern than yours.

What happens when you do this?

more /proc/cpuinfo | grep flags

eerobert, 2019-07-02:
step 7

eerobert, 2019-07-03:
[image]

wilderfield, 2019-07-03:
Your machine doesn’t support AVX2 instructions. I’m guessing that’s the
problem. Let me compare with the build system tomorrow. Bryan.

eerobert, 2019-07-03:
i also suspect that's the problem as i was running dnndk3.0 for some edge projects and facing the same problem.

wilderfield, 2019-07-03:
This is the machine that I build the docker image on, building the docker image, means I'm building our caffe from sources, as well as the xfdnn library "libxfdnn.so". Note that the build machine does support AVX2. At the moment, I am not sure how I can support older machines. Can you tell me the output of "lscpu"?

flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb invpcid_single intel_pt ssbd ibrs ibpb stibp kaiser tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm mpx avx512f rdseed adx smap clflushopt clwb avx512cd xsaveopt xsavec xgetbv1 cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts md_clear flush_l1d