Xilinx/ml-suite #94: IOError: [Errno 2] No such file or directory: 'work/deploy.prototxt'

KimdongHak, 2019-08-01:
Hi

I am trying to benchmark example (refer to : 
https://github.com/Xilinx/ml-suite/blob/master/examples/caffe/Benchmark_README.md)

Also, I am trying to run as following command:
--> python run.py --prototxt /opt/models/caffe/bvlc_googlenet/bvlc_googlenet_train_val.prototxt --caffemodel /opt/models/caffe/bvlc_googlenet/bvlc_googlenet.caffemodel --prepare --output_dir work

on U200 Alveo Accelator Card.

### **But, I faced the following error**

```
*** Check failure stack trace: ***
I0801 09:17:15.521344   362 net.cpp:455] pool2/3x3_s2_pool2/3x3_s2_fixed_0_split -> pool2/3x3_s2_pool2/3x3_s2_fixed_0_split_3
Network: work/deploy.prototxt
GenerateCode: work/compiler
Weights: work/deploy.caffemodel
PngFile: None
ConcatStrategy: None
Strategy: all
ScheduleFile: None
DDR: 256
DSP: 96
Verbose: False
FromTF: True
Memory: 9
Phase: TEST
RankDir: BT
Start compiling work/deploy.prototxt

**************************************************
* BUILDING DATA FLOW GRAPH
**************************************************
Traceback (most recent call last):
  File "run.py", line 131, in <module>
    Compile(args["output_dir"])
  File "run.py", line 59, in Compile
    compiler.compile()
  File "/scratch/MLsuite/xfdnn/tools/compile/bin/xfdnn_compiler_caffe.py", line 198, in compile
  File "/scratch/MLsuite/xfdnn/tools/compile/optimizations/changeinplace.py", line 42, in read_file
IOError: [Errno 2] No such file or directory: 'work/deploy.prototxt'
```

Can anyone help me how to resolve this issue ? 
[image]

wilderfield, 2019-08-02:
Hi,

It looks like you missed a setup step.

To quantize a model for running on FPGA, you need a calibration set of
images. These images are run through the computational graph and statistics
are gathered regarding the activations at every node.

Since these are Imagenet models we just use the Imagenet validation set for
calibration.

Your screenshot says that it can’t find the images. So the quantizer
crashed, and the compiler doesn’t know not to run, then you get strange
error during compilation.

Just make sure the steps were followed to download the Imagenet validation
set. See examples/caffe/README.md

KimdongHak, 2019-08-02:
Hi, wilderfield.

I really appreciate to your reply.

But I already tried to convert a calibration set of images by 3 times.
I could not understand if some files are converted and others aren't.
Please refer to below attached figure.

And I also found another issue.
The model of the equipment that I was running the example code is Dell PowerEdge R720.
And the CPU of this equipment does not support the instruction set AVX 2.0.
Therefore, I think it will not work normally because of these problems.

So, I had one more question.
Do you know if there is another example that does not require AVX2.0?

Best Regards.

[image]