Xilinx/ml-suite #95: u280 support

scarrazza, 2019-08-05:
Hi,

is it possible to use the current code (`latest`) with the alveo u280?
If I understand correctly the overlaybins for this board are not available yet so examples like the tensorflow notebook will not work. Is it possible to compile locally the overlaybins for that board?

Thanks.

wilderfield, 2019-08-06:
It is not possible to build the overlays aka “xclbins” for U280 unless you have the encrypted RTL for our engine. The RTL has to slightly change on our side to support internal HBM. This work has been ongoing for a while now, and the finishing touches are on the software side. Will get back to you on timeframe.

scarrazza, 2019-08-06:
Thanks for the feedback.