Xilinx/ml-suite #104: u280 container.

azrael0fog, 2019-08-30:
https://www.xilinx.com/member/forms/download/eula-xef.html?filename=xilinx-ml-suite-ubuntu-16.04-xrt-2018.2-tensorflow-1.12.0-mls-1.5.tar.gz

I have just bought a u280 card, but i find that the docker container of tensorflow and caffe is only supporting u200 and u250,  and it is a 2018.2 version. Could you tell how to download a new docker container? For example 2019.1 version.

wilderfield, 2019-08-30:
Hey,

For ML Suite we don’t have support for U280 yet. We are working on it now.

Our software supports 2018.2 XRT APIs.

We are currently developing 2019.2 related code, but that will not go live
for a couple months.

azrael0fog, 2019-08-30:
Thanks for your reply.  Looking forward to the new version. ヾ(◍°∇°◍)ﾉﾞ