Xilinx/ml-suite #105: No DSA Package insalled in System in Docker mluser container

SRQ91, 2019-09-04:
Hi,

I run the xbutil scan flash after opening the mluser docker container and get the following result:
`mluser@xxx:/opt/xilinx/xrt/bin$ sudo ./xbutil flash scan
Card [0]
	Card BDF:		0000:01:00.0
	Card type:		u200
	Flash type:		SPI
	DSA running on FPGA:
		xilinx_u200_xdma_201830_2,[TS=0x000000005d1211e8],[BMC=4.2.0]
	DSA package installed in system:	(None)`

Meanwhile, If I don't do this within the container I get:

`abc@xxx:/opt/xilinx/xrt/bin$ sudo /opt/xilinx/xrt/bin/xbutil flash scan
[sudo] password for abc: 
XBFLASH -- Xilinx Card Flash Utility
Card [0]
	Card BDF:		0000:01:00.0
	Card type:		u200
	Flash type:		SPI
	Shell running on FPGA:
		xilinx_u200_xdma_201830_2,[TS=0x000000005d1211e8],[SC=4.2.0]
	Shell package installed in system:	
		xilinx_u200_xdma_201830_2,[TS=0x000000005d1211e8],[SC=4.2.0]`

Am I doing something incorrectly while launching the docker container?

SRQ91, 2019-09-04:
UPDATE:
I copied the `xilinx-u200-xdma-201830.2-2580015_16.04.deb` into the container and installed it. Then I get matching DSA packages within the container:
`mluser@xxx:/opt/xilinx/xrt/bin$ sudo ./xbutil flash scan
Card [0]
	Card BDF:		0000:01:00.0
	Card type:		u200
	Flash type:		SPI
	DSA running on FPGA:
		xilinx_u200_xdma_201830_2,[TS=0x000000005d1211e8],[BMC=4.2.0]
	DSA package installed in system:	
		xilinx_u200_xdma_201830_2,[TS=0x000000005d1211e8],[BMC=4.2.0]
`
However if I try to run the yolov2 jupyter notebook, I get:
`I0904 13:03:07.065441   770 net.cpp:268] data does not need backward computation.
I0904 13:03:07.065444   770 net.cpp:310] This network produces output layer31-conv
I0904 13:03:07.065469   770 net.cpp:330] Network initialization done.
[XBLAS] # kernels: 1
CL_PLATFORM_VENDOR Xilinx
CL_PLATFORM_NAME Xilinx
CL_DEVICE_0: 0x359f850
CL_DEVICES_FOUND 1, using 0
loading /opt/ml-suite/overlaybins/alveo-u200/overlay_4.xclbin
Xclbin does not match DSA on card.
Please run xbutil flash -a all to flash card.
timestamp of ROM not match Xclbin
ERROR: Failed to load xclbin.
ERROR: Failed to create compute program from binary -44
Path work/deploy.caffemodel_data.h5 is a file.
Loading weights/bias/quant_params to FPGA...
[I 13:03:12.563 NotebookApp] KernelRestarter: restarting kernel (1/5), keep random ports
WARNING:root:kernel c0449a42-1ccf-4e8f-9e26-982eda84535f restarted

[I 13:03:52.873 NotebookApp] Saving file at /object_detection_yolov2.ipynb
`
How to resolve this problem? Also, I expected the container to have a DSA already available. Am I setting it up incorrectly?

SRQ91, 2019-09-04:
Although I though the DSA is fully updated, I still followed the error message and tried to update. As expected, flashing is not possibe:
`mluser@ebvprogsolution:/opt/xilinx/xrt/bin$ sudo ./xbutil flash -a all
Probing card[0]: DSA on FPGA is up-to-date
0 Card(s) flashed successfully.
`

matlinsas, 2019-11-16:
I met the same issue.
Any updates on this issue??