Xilinx/ml-suite #120: Couldn't check accuracy of quantized tensorflow model

katherineyun, 2019-10-28:
Hi, I am currently working with a custom tensorflow model. So far, the quantization was successful, but I would like to know the accuracy of quantization results. 

The following command is for testing quantized caffe network:
decent_q test -model quantize_results/quantize_train_test.prototxt -weights quantize_results/quantize_train_test.caffemodel  -test_iter 100

But I have encountered this error (running decent_q test in tf container):

decent_q: error: argument command: invalid choice: 'test' (choose from 'quantize', 'inspect', 'dump')

Also, what command I should use to test the accuracy of a quantized tensorflow network?

Thesane, 2019-10-28:
you can use the validate script
[https://github.com/Xilinx/ml-suite/tree/master/examples/tensorflow](https://github.com/Xilinx/ml-suite/tree/master/examples/tensorflow)

> python run.py --validate --model /opt/models/tensorflow/inception_v1_baseline.pb --output_dir work --input_nodes data --output_nodes loss3_loss3 --c_input_nodes data --c_output_nodes loss3_loss3 --input_shapes ?,224,224,3`

katherineyun, 2019-10-29:
> you can use the validate script
> https://github.com/Xilinx/ml-suite/tree/master/examples/tensorflow

Hi Ahmed, thank you for your answer! Do you know if there is a way to check the quantization accuracy without deploying the network onto FPGA?