Xilinx/ml-suite #123: After testing ，I found pool performance on the fpga-card，is that normal？

MCC1MCC, 2019-11-29:
at first I test yolov2-608, and found that U200 performance is lower than NVIDIA 2080Ti

and then I change the input image size for yolov2,  
then made the form，

[image]

[image]

U200 inference speed will only better when the input image size is lower than 224，

is that normal？ 

thanks！

MCC1MCC, 2019-11-29:
输入层尺寸 means input image size
推理用时 means inference time(ms/img)

zyxcambridge, 2019-12-01:
> at first I test yolov2-608, and found that U200 performance is lower than NVIDIA 2080Ti
> 
> and then I change the input image size for yolov2,
> then made the form，
> 
> [image]
> 
> [image]
> 
> U200 inference speed will only better when the input image size is lower than 224，
> 
> is that normal？
> 
> thanks！

你这边没有用docker 跑起来了，是吧

MCC1MCC, 2019-12-02:
> 你这边没有用docker 跑起来了，是吧

我用的是xilinx官方的docker，镜像名称是xilinx-ml-suite-ubuntu-16.04-xrt-2018.2-caffe-1.0-mls-1.5
请问你测试的推理速度如何？

zyxcambridge, 2019-12-02:
> 我用的是xilinx官方的docker，镜像名称是xilinx-ml-suite-ubuntu-16.04-xrt-2018.2-caffe-1.0-mls-1.5
> 请问你测试的推理速度如何？

我也用的是xilink的官方docker xilinx-ml-suite-ubuntu-16.04-xrt-2018.2-caffe-1.0-mls-1.5 在 aws f1 上 ，但是各种环境不兼容；

请教一下：  你这边是本地机子上 装了一块 u200的 FPGA卡，才跑起来的是吧；

另外我这边 想知道一下  yolov2 和yolov3的MAP ，感谢，
顺便来个微信 吧，zhangyixin395104  我的微信

llt19903767731, 2020-03-19:
> 我用的是xilinx官方的docker，镜像名称是xilinx-ml-suite-ubuntu-16.04-xrt-2018.2-caffe-1.0-mls-1.5
> 请问你测试的推理速度如何？

这个镜像是XRT 2018.2版本的吧，如果本机装Xilinx的XRT是2019年的，好像没有对应的镜像吧？我在仓库好像没找到。