Xilinx/ml-suite #127: Alveo U50 ml-suite  Unknown: exceptions.RuntimeError: Could not init FPGA: xclbin

jaydevshelat, 2020-04-15:
Hi,

I am using Alveo U50 board.  I am trying to run the ml-suite tensorflow example. I have tried command line as well as Jupyter notebook.  I get the same error (show below).

My board is powered-up & initialized.  One observation is the utilities (for e.g. docker_run.sh script) don't recognize my board/platform automatically.  

I am fairly sure this is because the docker container does not have an overlaybin for U50. 

Does anyone know if this works on U50?  If yes, where to get the correct docker container?

Error received:
`...
save graph at ./inception_v1_baseline-fpga.pb
[XBLAS] # kernels: 1

ERROR: No devices found
ERROR: Failed to find an OpenCL platform
2020-04-15 22:35:22.928304: W tensorflow/core/framework/op_kernel.cc:1261] Unknown: exceptions.RuntimeError: Could not init FPGA: xclbin /opt/ml-suite/overlaybins/(unknown)/overlay_4.xclbin lib_path /opt/ml-suite/xfdnn/rt/libs/libxfdnn.so
Traceback (most recent call last):

  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/script_ops.py", line 206, in __call__
    ret = func(*args)

  File "/opt/ml-suite/xfdnn/rt/xdnn_rt_base.py", line 134, in forward_exec
    layer_outputs = layer.forward_exec( layer_inputs )

  File "./xfdnn/tools/emu/fpga_pydot_layer.py", line 103, in forward_exec

  File "/opt/ml-suite/xfdnn/rt/xdnn_env.py", line 83, in createHandle
    raise RuntimeError("Could not init FPGA: xclbin %s lib_path %s" % (self._xdnnParams['xclbin'], self._xdnnParams['lib_path']))

RuntimeError: Could not init FPGA: xclbin /opt/ml-suite/overlaybins/(unknown)/overlay_4.xclbin lib_path /opt/ml-suite/xfdnn/rt/libs/libxfdnn.so

Traceback (most recent call last):
  File "run.py", line 134, in <module>
    top5_accuracy(graph, args.input_nodes, args.output_nodes, iter_cnt, batch_size, args.label_offset)
  File "/opt/ml-suite/examples/tensorflow/utils.py", line 106, in top5_accuracy
    predictions = sess.run(output_tensor, feed_dict={tensor: inputs[name] for name, tensor in input_tensors.items()})
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/client/session.py", line 929, in run
    run_metadata_ptr)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/client/session.py", line 1152, in _run
    feed_dict_tensor, options, run_metadata)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/client/session.py", line 1328, in _do_run
    run_metadata)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/client/session.py", line 1348, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.UnknownError: exceptions.RuntimeError: Could not init FPGA: xclbin /opt/ml-suite/overlaybins/(unknown)/overlay_4.xclbin lib_path /opt/ml-suite/xfdnn/rt/libs/libxfdnn.so`

wilderfield, 2020-04-15:
Alveo U50 is only supported in the Xilinx/Vitis-AI github repository.

Vitis-AI is evolved from MLSuite and has a broader scope. It supports more
boards, plus edge devices.

https://github.com/Xilinx/Vitis-AI

jaydevshelat, 2020-04-21:
Thank You @wilderfield !  
**One Suggestion:** It would be nice to have target platform in the readme.  Even in Vitis-AI it looks like U50 is not well supported (compared to U200/U250/ZCU boards).  Again, having this information in different readme links would be of great help to a Xilinx beginner!